tszheichoi/awesome-sensor-logger #10: Run always in background

moltra, 2023-10-16:
I would love for this app to run all the time on my Android Phone.  I would like it to save the current days readings and start new files for the new day.  
I am going to see if I can get IFTTT to start it automatically and use the http server to export it daily. 
I am also want a button to log a timestamp and a pain level between 0-10.

I am going to try and create an app like this on my own, but I have never created an android app.

tszheichoi, 2023-10-19:
When it comes to logging a timestamp for pain level, you can use Sensor Logger's existing annotation feature. During a recording session, simply tap on the pencil icon located next to the Stop Recording button, and enter the pain level. This will be stored along with the timestamp in the `Annotation.csv` file in the recording.

Regarding wanting to launch the app and start a recording automatically, this is not a feature of Sensor Logger. I'm concerned there are limitations on Android or iOS that might make such a feature challenging to implement or use, but have not looked in detail. Let me know the details if you discover that this is possible.

The idea of splitting an ongoing recording into separate daily files, over a continuous recording session spanning multiple days is also not currently supported, although this is something I'd like to add to the app.

moltra, 2023-10-20:
Thank you, I am going to try using the pencil icon method.

closed 2023-10-20 by moltra