tszheichoi/awesome-sensor-logger #12: Can the app stream audio through the watch?

yujie-tao, 2023-10-21:
I tried to start the recording from my Apple Watch but it looks like the output file still recorded through my phone.

tszheichoi, 2023-10-21:
Sensor Logger currently does not support audio recording on the watch. It is a feature on the roadmap, but there is currently no timeline for this due to the technical complexity involved.

closed 2023-10-21 by tszheichoi

yujie-tao, 2023-10-21:
Thanks for the quick response! As an engineering student myself looking to prototype this, I am curious: what would you say the major technical complexity is? Do you happen to know any open-source project trying to do similar things?

tszheichoi, 2023-10-21:
Sensor Logger is designed where the iOS app creates all the recording files its-self on device. I think the way that watch audio recording would be implemented given the API restrictions is that the audio recording would be triggered on the watch side to a new file on the watch, and then that file would get sent over to the phone in the background at the end of the recording to the iPhone.
For Sensor Logger, this would add quite a bit of complexity (what happens if you export the recording before the audio file has arrived etc)

yujie-tao, 2023-10-22:
Thanks for sharing the insights!