tszheichoi/awesome-sensor-logger #13: Uncalibrated Values cannot be saved on Google Pixel 4a

takanotume24, 2023-10-31:
Hello.

I am using Sensor Logger on a Google Pixel 4a.
In `Settings`, even though I have `Uncalibrated Values` turned on, the Uncalibrated data for magnetics, acceleration and gyro are not saved.

To be precise, when I export the zip file after recording, `AccelerometerUncalibrated.csv`, `GyroscopeUncalibrated.csv` and `MagnetometerUncalibrated.csv` are generated. but the content of the file is 0 bytes.
Also, when I check the recorded data from the `Recording` list, the Uncalibrated data does not show the time series graph, while the Calibrated data shows it correctly.
This phenomenon occurs on several Google Pixel 4a units. (So I don't think any particular device is broken.)

However, Google Pixel 6 seems to be able to save Uncalibrated data correctly.

Finally, here are the respective versions.
- Google Pixel 4a
  - OS: Android 13, TQ3A.230805.001
  - SensorLogger: 1.24.0 Build 3145851

- Google Pixel 6
  - OS: Android 13, UP1A.231005.007
  - SensorLogger: 1.24.0 Build 3145851

If there is any other information you need to help me solve the problem, please let me know. 
Thanks for the great app!

tszheichoi, 2023-11-06:
Hi @takanotume24, thanks for reporting this issue. 

I have found a bug in the Sensor Logger code and this issue will be fixed in the next release 1.25.0. This should be available in the coming days.

takanotume24, 2023-11-07:
Hello! Thanks for the quick fix. I will check when the update is delivered to my device.

tszheichoi, 2023-11-12:
Hi @takanotume24, version 1.25.0 has been out for a few days now. Has it resolved the issue? If not, may reopen the issue.

closed 2023-11-12 by tszheichoi

takanotume24, 2023-11-21:
@tszheichoi Hello! Thanks for fixing it so quickly!
I confirmed that the Uncalibrated Data is saved on the Pixel 4a, I can't check on the Pixel 5a as I don't have it with me right now, but it will probably be fine.

Thanks! I will report back if there are any more problems.