tszheichoi/awesome-sensor-logger #14: [Support / Analysis] Trying to map a path based on GPS start coordinates, but continueing only with sensor data.

mcdope, 2023-11-19:
Heya,

This is not a bug report or like that, but rather a request for tips how to do what I want to do. I have a hard time wrapping my mind around it.

**The scenario:** I stroll around somewhere, having GPS signal, and then go underground and losing at least GPS, most likely even cellular etc. All I have is the data recorded by "Sensor Logger". Could take additional paper notes / annotations thought. When back at home I want to take this data and plot a path like a polyline / line based on multiple points (at least thats the easiest way I think). The idea is to just walk around and get a rough map out of the recording at the end. I want to create a script where I can drop in the combined CSV from the app, process the data (that's the part I need help with) and draw the result (easy, no help needed).

**Usecase:** Exploring a tunnel network below a historic industry place and mapping it. Since I don't care too much about accuracy, but more about a general overview and orientations, I guess this should be not that hard. 

What I don't get is: how can I do this exactly, or more detailed: what are the best sensors to use for that? And how to adjust/convert them to relate them to the gps coords from the starting point? 

**Sidenote:** Looking at https://www.tszheichoi.com/coaster it seems like its basically what I'm looking for, at least regarding the resulting view.

tszheichoi, 2023-11-20:
Hi @mcdope, sounds like an interesting and challenging problem. The general problem you are describing is related to indoor positioning system. See https://en.wikipedia.org/wiki/Indoor_positioning_system, particularly the section on non-radio technologies. The problem becomes a lot easier too if you have Bluetooth or wifi beacons to help with the localisation -- but presumably, that is not the case. 

You pointed to my roller coaster data fusion project, which does indeed address a similar problem. Will be happy to have a chat and learn more about the specifics of your use case before jumping to any technical solution. Feel free to reach out on sensorlogger@tszheichoi.com and we can go from there. If you have some sample recordings you have made with Sensor Logger so we can investigate further, that will be even better!

tszheichoi, 2023-11-24:
Closing this now -- please reach out as stated in my previous comment.

closed 2023-11-24 by tszheichoi