tszheichoi/awesome-sensor-logger #19: collect data from airpods

18805120199, 2024-01-10:
I wear airpods and use iPhone to collect data. However, in the exported csv file, there is no data in 'Headphone.csv'. How can I collect data from airpods?

18805120199, 2024-01-10:
> I wear airpods and use iPhone to collect data. However, in the exported csv file, there is no data in 'Headphone.csv'. How can I collect data from airpods?

The data I need is the acceleration and angular velocity data of airpods.

tszheichoi, 2024-01-12:
Only some AirPod models support headphone motion tracking on iOS. The easiest way to identify this is to check whether your model supports Spatial Audio. You can find more information on Apple’s documentation: https://support.apple.com/en-us/102469. Further you should ensure that your AirPod is paired properly with the phone you are running Sensor Logger on.

closed 2024-01-12 by tszheichoi

18805120199, 2024-01-17:
Thank you very much. I still have two questions. One is: does the
calibrated accelerometer data collected by the sensor logger include
gravity acceleration? The second one is what is the difference between
calibrated and uncalibrated data of accelerometer, gyroscope and
magnetometer?

tszheichoi, 2024-01-17:
The Accelerometer Sensor does not include gravity (this can be recorded separately via the Gravity sensor).

The calibrated sensors have undergone adjustments during manufacturing to ensure accurate measurements, the specific changes depend on phone model and specific sensors - I recommend recording both to see the different on your device.