tszheichoi/awesome-sensor-logger #20: iPhone 14 Pro Max wrong axes

Josefer25, 2024-02-08:
I've read [apple sensor documentation](https://developer.apple.com/documentation/coremotion/getting_raw_accelerometer_events) which was mentioned in a similar issue. The problem is that X and Y axis are inverted, If I move my phone to the right, the accelerometer is negative until I stop that is positive. The same happens with the axis Y and Z. This produces wrong results for the uncalibrated accelerometer, as the gravity is negative but it should be positive to be consistent, so methods like Magdwick are failing.

Am I missing something?

Josefer25, 2024-02-08:
If I throw my phone to the ground (padded ofc), the maximum Z axis acceleration is negative (which shouldn't be, should be contrary to gravity)

tszheichoi, 2024-02-24:
Apologies for the delayed reply, you are correct that the axes appear inverted when compared to the picture on Apple's documentation. All you need to do is to apply a negative sign to all three axes. Apple's definition of acceleration seems to be in the context of inertial force, which is the opposite of accelerating force. 

I have spent some time producing comprehensive documentation on the coordinate system, and more importantly, how it may differ across iOS and Android -- this is something a few users have recently asked for. Take a look and see if it makes sense / is consistent with your experimentation https://github.com/tszheichoi/awesome-sensor-logger/blob/main/COORDINATES.md

tszheichoi, 2024-02-25:
New in version 1.29 of Sensor Logger is a coordinate system harmonisation option. This will convert the measurements on iOS to the more intuitive and familiar coordinate system, in line with Android. 

For more information see https://github.com/tszheichoi/awesome-sensor-logger/blob/main/CROSSPLATFORM.md.

closed 2024-02-25 by tszheichoi

Josefer25, 2024-02-25:
That is amazing mate. I had problems with the gravity as I said which seemed an error and not just an axis inversion.

I'll try tomorrow but your documentation and the new option is quite useful. Thanks again!