tszheichoi/awesome-sensor-logger #21: Allow Creation of Studies with Sensors that are not on the Investigator's Phone

stcz, 2024-02-22:
Thanks for creating this nice app.

I'm currently testing the features before buying it, almost everything looks good.

On thing i found is an issue for me.

I have multiple devices. On one device I want to create the study. This device doesn't have a Barometer. But i also want to include the Barometer on devices that have one.  Is it currently possible to do this?

Might it be a better solution to show always all Sensors, when create a study as they are not recorded at all, if they not exist on a device?

Thanks for your answer.

tszheichoi, 2024-02-22:
Thanks for your interest in Sensor Logger. Supporting sensors in studies that are not supported on the creation device is a sensible feature, and should be easy to implement from my end. Expect this to be possible in the next release of the app. I will update this thread once it is available.

tszheichoi, 2024-02-25:
Version 1.29 is now available on both app stores, and support enabling sensors that are not available on the Study creation device.

closed 2024-02-25 by tszheichoi