tszheichoi/awesome-sensor-logger #28: Feature Request: Allow to delete data in the cloud

stcz, 2024-03-06:
As sometimes data is not usable (not because of the app) it would be great, if the investigator cloud delete data from the cloud to free space. I would really appreciate this feature. It is also fine to make the feature to a Plus or Pro feature.

tszheichoi, 2024-03-06:
This is now possible as of about a week ago. You should see a delete button next to each recording when browsing them as an investigator on the study's site. 

[image]

closed 2024-03-09 by tszheichoi

stcz, 2024-03-09:
You are right, I investigated that about two weeks ago. Thanks for the great Maintenance of the App!