tszheichoi/awesome-sensor-logger #29: Feature Request: Keep awake with proximity sensor to save battery

stcz, 2024-03-06:
As we all know, it is difficult for many smartphone vendors to let tasks run in background the recordings typically have to run in Keep Awake mode.

A suggestion (from [here](https://github.com/phyphox/phyphox-android)) it would be great to have the possibility to lock the phone and turn the screen off, when the proximity sensor is triggered (as during a phone call). So it would be possible to keep the app awake, with display off, and put the phone into your pocket without randomly touching on the screen. 

I would be very happy to see this feature and I think this will be a real improvement for the app.

tszheichoi, 2024-03-06:
Very interesting, thanks for bringing this to my attention. Seems possible, but I have to play around with it more to better understand how this will interact with various Sensor Logger functionalities. Will keep this thread updated once I have investigated a bit more.

tszheichoi, 2024-03-09:
I can now confirm that this option will be available in version 1.30.

stcz, 2024-03-09:
Thanks, that will be great :heart:

tszheichoi, 2024-03-10:
Version 1.30 is now available. Do re-open this issue if you run into further issues.

closed 2024-03-10 by tszheichoi