tszheichoi/awesome-sensor-logger #34: GNSS observables

kcleineweber, 2024-04-23:
Is it possible to log not only the location solutions but the raw observables - pseudorange, doppler, adr? Thanks

tszheichoi, 2024-04-24:
Sensor Logger currently does not support these raw values, primarily because it is not available on iOS, and I want most features to be reasonably cross-platform. See, for example, https://developer.apple.com/forums/thread/693229. On the other hand, I believe it is technically possible on Android. Are there particular reasons / use cases you want to achieve with these? If so, happy to consider implementing.

tszheichoi, 2024-05-05:
Closing this for now. If you have specific needs for raw GPS values on Android, then do reach out again.

closed 2024-05-05 by tszheichoi