tszheichoi/awesome-sensor-logger #35: Feature Request: Allow to re-enable a study

stcz, 2024-04-24:
Thanks again for the great app. I wanted to ask for a new feature:

As I forgot to extend the Duration of a study, i now need to create a new one with the same settings, what is a lot of work.
It would be great if there would be some period of e.g. a week when you are able to re-enable a existing study again.

tszheichoi, 2024-04-24:
In the next release of Sensor Logger, there will be a duplicate Study feature, so there will be a lot less work in terms of creating a new one with the same settings. As for the grace period idea, it makes sense in principle, but I will need to have a think in terms of the implementation.

tszheichoi, 2024-05-05:
Closing this now as the Study duplication feature is now available. Study extension post-ending is not something that I will get to implement in the near future due to other priorities on the roadmap.

closed 2024-05-05 by tszheichoi