tszheichoi/awesome-sensor-logger #36: Wrist Orientation

abcamiletto, 2024-04-30:
Hi!

I've been using your app and I am quite a fan so far.
What I have been noticing is that the wrist orientation is very snappy

[image]

It looks like this happens at regular intervals, and it's only happening in the watch orientation, not in the phone one.
Any idea on what might cause this?

Thanks a lot in advance!

tszheichoi, 2024-04-30:
I will first just double check these tips and tricks, and ensure the relevant permissions have been set: https://www.tszheichoi.com/sensor-logger-watch-reliability

abcamiletto, 2024-04-30:
Thanks a lot for your comment. I think it's going in the right direction.
I checked, and I already had the suggested permission enabled.

What I realized is happening is the following:

  - If the wrist movement is slow and pointed to the user, everything is fine (the logger app is always open on the watch and I can see the timer)
  - If the wrist movement becomes faster and does not point to the user's face anymore, the watch screen gets black, and the iPhone app signals "Watch Disconnected" even for just a moment. It looks like some gesture recognition is putting the watch to sleep. It then wakes up very shortly after, and this keeps happening in a cycle every 5s.

Although I must say, I saw at least a couple of occasions when this 5s reset was present, but no disconnection was signaled.

I am currently using an out-of-the-box Apple Watch SE, if this matter.
Thanks a lot for answering in here!

abcamiletto, 2024-05-01:
Quick update as I have pinpointed two separate problems

- Sometimes, when the watch is not facing the user, it will turn the screen black and disconnect from the phone, stopping to send data. If the watch wakes up, it will restart sending data. This happens around 30% of the times
- Most of the time, even if the screen turns black, the watch keeps on sending data to the iPhone. In these cases, there is a substantial jump in the orientation (obtained via "wrist motion" -> "quaternionX, quaternionY, quaternionZ, quaternionW" every 5s flat, with remarkable accuracy (it's always 5s +- 0.01s).

The settings are the ones suggested in the guide (I did not have to change them; they were set like that by default), so I'm not sure what could be the reason behind these.

Thanks a lot for the patience!
Andrea

bchandaka, 2024-05-05:
I found starting a workout session on the watch, and then switching to the sensor logger app and starting the recording allows the watch to continue recording data when the screen is off. 
I think this is mentioned in the FAQ section, "My watch keeps disconnecting" here: https://www.tszheichoi.com/sensorloggerhelp

tszheichoi, 2024-05-21:
Just to update this thread that the reliability improvements to the watch companion app is still being worked on, and I will update when I have more to share. In the mean time, do reference this documentation: https://www.tszheichoi.com/sensor-logger-watch-reliability

tszheichoi, 2024-06-06:
Version 1.35 is now out with a revamped Apple Watch connection mechanism. This should resolve a lot of reliability issues.

tszheichoi, 2024-06-09:
Closing this now. Feel free to re-open if you encounter further issues with the latest version of Sensor Logger

closed 2024-06-09 by tszheichoi