tszheichoi/awesome-sensor-logger #37: Delay in Apple Watch data Streaming over HTTP

bchandaka, 2024-05-05:
Hi,

I'm using sensor logger to stream data on my iphone, airpod, and apple watch over http to a python flask server which saves and visualizes this data. For the apple watch however, there seems to be a 5-10 second lag in when the data arrives to my server. The timestamps still matches up with other correlating phone data(moving the phone and watch together at the same timestamp has similar acceleration jump etc.). Usually, I just wait a couple extra seconds before ending the recording to make sure the watch data gets caught up a bit.

 Is there any cause for why the watch data might be delayed when streaming over http?

tszheichoi, 2024-05-21:
Hi @bchandaka could you clarify whether the delay only exists for the stream? Is there any delay / offset / data issue with the local recording on the phone when you expect it after the fact?

tszheichoi, 2024-06-06:
@bchandaka Version 1.35 is now out with a revamped Apple Watch connection mechanism. Please try again if still interested.

tszheichoi, 2024-06-09:
Closing this now. Feel free to re-open if you encounter further issues with the latest version of Sensor Logger

closed 2024-06-09 by tszheichoi