tszheichoi/awesome-sensor-logger #44: App is regularly stopped by WebView updates

arneschuldt, 2024-08-13:
# Background

In order to facilitate continous operations of the app, we are taking the following precautions:

* Prevent the app form being stopped if it is not "actively used" (Android app configuration)
* No restriction in battery use (Android app configuration)
* Prevent automatic updates (Google Play configuration)

# Problem

However, there is an issue with the Android WebView component the app uses. In the logs, we see that the sensor app is regularly stopped due to WebView updates, e.g.:

```
08-13 06:27:33.409  1053  1122 I ActivityManager: Killing 12357:com.kelvin.sensorapp/u0a283 (adj 200): stop com.google.android.webview due to installPackageLI
```

The reason seems to be that Google reserves the right to fix critical vulnerabilities (in the WebView in this case) even if updates are disabled:
> If Google determines that an app update fixes a critical security vulnerability, we may make certain app updates. These updates can occur regardless of update settings in the app or on your device.

source: https://support.google.com/googleplay/answer/113412?hl=en

# Possible Solution

For its core functionality (collection sensor data), the sensor app does probably not need the WebView. We assume that the WebView is mainly used to show web content, e.g., the "about" section of the app.

If that is true, one option might be to remove the WebView component and open such information directly in the system web browser (e.g., Chrome). This should hopefully increase the app stability as it then no longer depends on Google's WebView security updates.

tszheichoi, 2024-08-13:
Thank you for the detailed report on this issue. I have looked into it and this is caused by the internal implementation of the plots in Sensor Logger which use a Web View.

I will add an optional setting to disable plots in the "Recording & Workflow" Settings page in the August release of Sensor Logger, which is coming soon.

tszheichoi, 2024-08-19:
@arneschuldt Sensor Logger version 1.37 is now out with this new option under the Recording & Workflow section. Please give that a try and see if that fixes the issue. I will close this now, but feel free to reopen if you run into problems.

closed 2024-08-19 by tszheichoi

arneschuldt, 2024-08-19:
@tszheichoi Thanks a lot.

We will continue our testrun and report back in case of any problems detected.

arneschuldt, 2024-10-07:
@tszheichoi

It seems as the problem is actually solved. It did not occur again. Thank you very much!