tszheichoi/awesome-sensor-logger #45: Headphone Motion Data Not Recorded When Using Both Headphone Motion and Audio Simultaneously with AirPods Pro

supernaiter, 2024-08-20:
#### Environment
- **Device**: iPhone 12
- **Earphones**: AirPods Pro (2nd Generation)
- **Software Version**: iOS [Specify Version]

#### Steps to Reproduce
1. Connect AirPods Pro to the iPhone 12.
2. Enable the Headphone Motion setting.
3. Enable the Audio setting.
4. Perform actions or movements that should be recorded by the Headphone Motion sensors.
5. Attempt to retrieve or verify the Headphone Motion data.

#### Expected Behavior
Both Audio and Headphone Motion data should be recorded and retrievable simultaneously without any data loss.

#### Actual Behavior
When both Headphone Motion and Audio settings are enabled, the Headphone Motion data is not recorded or retrievable.

#### Additional Information
It appears that enabling the Audio setting interferes with the recording of Headphone Motion data. This issue is critical for applications that require simultaneous use of both features.

#### Impact
This issue impacts applications that require precise movement data from the Headphone Motion sensors while also using audio features, particularly for use cases like silent speech recognition, motion monitoring, and other audio-visual integrations.

#### Request
Please investigate this behavior to determine if it is a bug or by design. If it is a bug, a fix would be highly appreciated as it affects the capabilities of applications relying on both audio and motion data simultaneously.

---

tszheichoi, 2024-08-22:
Thanks for the detailed report -- I have been able to reproduce the issue on my end. I will investigate a fix and update when I can.

supernaiter, 2024-08-23:
FYI, if you disable the "Automatic Ear Detection" feature, the AirPods will continue to operate even if they are not in your ears, effectively functioning as standalone wireless accelerometers.

tszheichoi, 2024-08-30:
Just an update -- I have determined that this is a limitation/bug on Apple's API. I have reached out to Apple to understand more.