tszheichoi/awesome-sensor-logger #51: CoreML model trained with exported CSV does not classify correctly when raw data is passed from Apple Watch

grenos, 2025-01-13:
Hi, I have a trained a model with Apple's CreateML using wrist motion data recorded directly from the companion watch app as a standalone recording (not through the phone).

On testing and previewing the model inside the CreateML app using files exported from the the sensor logger app everything works great. But later, on my app passing raw data from the watch to the model I always get incorrect predictions. 

I just wanted to understand if the data exported from the sensor logger have any kind of processing done by default? 

I had a look at the [UNITS](https://github.com/tszheichoi/awesome-sensor-logger/blob/main/UNITS.md#wristmotion) readme but it's not really clear to me if maybe there is any processing that I need to do on my raw data.

I have data Standardisation OFF on the settings.

davidecarpini, 2025-01-13:
same issue here guys ➕

grenos, 2025-01-13:
my bad, it was a mistake on my end. Closing.

closed 2025-01-13 by grenos